circuitpython/web-editor #10: with microbit V2 it is not working on windows nor mac

apws, 2021-12-03:
hi I tried to post this to forum topic about microbit V2, but when submitting some f... crazy thing stopped me becasue of "CMSIS-DAP" word but then it was also impossible to submit after edits, I am starting to hate entire microbit thing deeply....

I tried microbit V2 (CP 7.1.0 beta 1 and nightly build too) on windows and also on mac but its frustrating - on mac I was able to go to editor once but then it hanged after first save&run too (actually then impossible to shutdown mac because of chrome, I must force switch it off).

On windows in chrome I tested chrome://bluetooth-internals, where lots of "Unknown or Unsupported Devices with different MAC started to occur - this also happened once from webeditor in "pairing dialog" - stream of unsupported devices after powering up microbit. Also on windows I see immediately after power-up in connected devices (so without any pairing activation with reset during fast yellow leds) device "BBC microbit CMSIS-DAP." and when I try to add, CIRPY shows as possible. (Here I mostly don't paired and removed also the CMSIS-DAP thing before trying webeditor connecting)

On Mac, I am now trying also chrome://bluetooth-internals and here is CIRCUITPY531b device but trying to inspect it from here to show services, means rotating circle only (and just after inspect, it was red message unsupported device also - the same trying to connect). On CIRPY device services are inspectable here. Now after trying to Pair CIRPY from webeditor, I see again only blue animated gif and latest chrome asi again totally hung.

makermelissa, 2021-12-03:
With windows, you have to add it to your system bluetooth devices first, otherwise it just won't work. I've heard it works with Edge browser too, but haven't tested that myself. I'll have to check on Mac. It sometimes takes a few tries to get it initially connecting, but is usually pretty stable once you do get it working.

makermelissa, 2021-12-03:
I got it working fine on mac. I was unable to get it working on either chrome nor edge on windows.

apws, 2021-12-03:
thanks for info... I have here similar format board with rp2040, so will try this (maybe add to boards)

idreamsi, 2021-12-04:
> [image] hi I tried to post this to forum topic about microbit V2, but when submitting some f... crazy thing stopped me becasue of "CMSIS-DAP" word but then it was also impossible to submit after [...]

I have the same problem on Windows 10.

tannewt, 2021-12-06:
@beaufortfrancois may be interested in following along.

makermelissa, 2022-11-23:
There has been a major rewrite of the bluetooth component. As such, many error messages have changed. Issue #40 has some more recent error messages, so I'm closing this in favor of that issue as it should be easier to spot. If it turns out to be unrelated, I can always reopen this issue.

closed 2022-11-23 by makermelissa