circuitpython/web-editor #12: Upload Files?

kyleplo, 2021-12-19:
Is there a way to upload files to my CircuitPython device using this web editor? If so, it should be easier to find. If not, it should be added.

tannewt, 2021-12-20:
You mean opening a local file and then sending it over?

makermelissa, 2021-12-20:
Not currently other than creating a new file and pasting in the text.

kyleplo, 2021-12-20:
Yes, I mean sending a local file. I wanted to do this for audio files, but this would also be useful for images and other binary files.

makermelissa, 2021-12-20:
I was thinking some file management utilities would be handy, but uploading would be as well and it could all be part of the same thing.

makermelissa, 2022-07-26:
@tannewt do you think https://github.com/adafruit/ble-file-transfer-js would handle binary files without modification?

tannewt, 2022-07-27:
> @tannewt do you think https://github.com/adafruit/ble-file-transfer-js would handle binary files without modification?

I'd like to think so but haven't tried it.

makermelissa, 2022-07-27:
Fair enough. I haven't tried either.

makermelissa, 2022-08-26:
I added code that works on Web Workflow and it _should_ work on BLE, though I can't seem to find my CLUE to test it. I'll stop by my office later and grab one to test with.

I _think_ it should work with how it is now, though there are extra parameters than JavaScript should just discard. If not, I have a fix for that in branch that also hasn't been tested.

makermelissa, 2022-08-26:
I tested and it caught a bug in the file transfer module. I created an issue here: https://github.com/adafruit/ble-file-transfer-js/issues/6. Also, it worked fine without accounting for the parameters.

makermelissa, 2022-09-28:
Fixed via https://github.com/adafruit/ble-file-transfer-js/commit/f85c66c2dec4b4019885eb3b7e2f4b8e3a4ab6a6 and 04dd38fa20344c85986b887272f8ff8133dff4f3.

closed 2022-09-28 by makermelissa