circuitpython/web-editor #15: Edit CIRCUITPY files

tannewt, 2022-06-22:
Use the directory picker to get access to a CIRCUITPY drive: https://developer.mozilla.org/en-US/docs/Web/API/HTMLInputElement/webkitdirectory

makermelissa, 2022-09-21:
Unfortunately I don't think this will work the way you may be thinking it will.

While it sort of works for selecting a folder with a couple of caveats (mostly the scary warning that it will upload a huge number of files from the path that is selected), the folder location isn't really useful as you can't specify where you would like to save files on a user's system. I mean you might be able to do a scan of the files on the device to get an idea of the layout, but since it is subject to change after that, this wouldn't be very useful.

For now, I think an approach where we open a single file and download a single file will work best unless we think of something that's a bit more clever that doesn't interfere with web security.

If I'm incorrect and there is a fancy way to do this, feel free to provide an example of another site that is able to work in a similar fashion. Closing for now.

makermelissa, 2022-10-19:
Reopening because this looks like using https://developer.mozilla.org/en-US/docs/Web/API/FileSystemDirectoryHandle might be a better approach, but basically the same idea.

PR #42 (2022-12-05): USB Workflow Initial Rewrite
This is a bit rough around the edges, but functional.

Fixes #15. Fixes #21.

Oh yeah, it also fixes an issue with xterm not resizing properly sometimes.
closed 2022-12-05 by tannewt