circuitpython/web-editor #20: Future Ideas for additional features

makermelissa, 2022-09-14:
This issue will mostly serve as a thread that other can contribute to with some ideas for future functionality with the CircuitPython Code Editor. Here's an initial list I came up with based on taking a look at https://python.microbit.org/v/beta:

- Some code snippets which could be cool. For snippets, perhaps either we could make use of Read The Docs or include some code snippets in the repository of the library and have some kind of interface where it grabs the code snippets we have there
    - We could also possibly use Read The Docs to display some integrated documentation
    - The tricky part would be having it know which docs went with specific library functions
- Maybe we could do something like Pyleap and load code from a learn guide in the future, though the details would need to be worked out.
- Adding some tabs and just displaying the active document in the serial terminal and editor
    - This would also be a bit tricky with juggling and transferring between URLs, but not impossible
- Adding additional language support
- Font sizing is a great idea
- Perhaps some color themes for people who don’t like the dark look of the editor
- Adding something like CircUp to the Editor as well like has been discussed in various places
- Deployment to Devices has been discussed
- Perhaps adding some kind of Self Signed Certificate generator may be an option along with deployment so we can do HTTPs for Web Workflow.

makermelissa, 2022-09-14:
Another idea I had was to support additional language color coding based on file extension that was opened. This would be fairly easy to add.

urfdvw, 2022-10-19:
In USB workflow.
Opening a folder instead of a file is going to make many interesting features possible.
For example, instead of creating a new file, download it and drag it into the CIRCUITPY drive.
Having a directory handle will allow you to directly create a new file on the CIRCUITPY drive edit it.
Many other functions like reading the CPY version number from boot_out.txt are also possible in that way.

makermelissa, 2022-10-19:
> In USB workflow. Opening a folder instead of a file is going to make many interesting features possible. For example, instead of creating a new file, download it and drag it into the CIRCUITPY [...]

Huh, ok. If you know how to to this, or could at least provide further information, that would be great. I tried to implement this using webkitdirectory (as I described in more detail in #15) and it just returned all of the files. Perhaps I didn't explore this option as thoroughly as I should have.

makermelissa, 2022-10-19:
Is this what you're referring to? https://developer.mozilla.org/en-US/docs/Web/API/FileSystemDirectoryHandle

urfdvw, 2022-10-19:
> > Is this what you're referring to? https://developer.mozilla.org/en-US/docs/Web/API/FileSystemDirectoryHandle
> 
> Ok, yeah, this looks very promising and is more in line with how I wanted USB workflow to work anyways. I will explore this option very soon. Thanks for the suggestion and pointing me in the right direction.

No problem.
Just beware that some of the features are only available in https environment.
And even in https I didn't get renaming file working. If you know how please also let me know.

makermelissa, 2022-10-19:
Thanks. The intention is to have the USB workflow only run from https://code.circuitpython.org, so #28 would probably need to be completed first.

urfdvw, 2022-10-19:
> Thanks. The intention is to have the USB workflow only run from https://code.circuitpython.org, so #28 would probably need to be completed first.

Sorry, just want to know a little more. What do you mean by "run from https://code.circuitpython.org/"?
Are the workflows currently running somewhere else?

makermelissa, 2022-10-19:
> Sorry, just want to know a little more. What do you mean by "run from https://code.circuitpython.org/"? Are the workflows currently running somewhere else?

Ah, to get around some browser security issues, Web Workflow loads from the device itself using http. The device loads https://github.com/circuitpython/web-editor/blob/main/assets/js/device.js which pulls loads the website. More information is available at https://learn.adafruit.com/getting-started-with-web-workflow-using-the-code-editor/usage.

makermelissa, 2024-02-23:
I think the circup-like feature (as mentioned in the first post above) is what would be probably most useful. With Web Workflow, because the file system needs to have USB mass storage disabled, updating libraries becomes really inconvenient.

makermelissa, 2024-02-23:
Something that could be built on top of the circ-up feature would be a code scanner that automatically detects the libraries you need and asks if you would like to update/install them.

urfdvw, 2024-02-24:
> Something that could be built on top of the circ-up feature would be a code scanner that automatically detects the libraries you need and asks if you would like to update/install them.

curious to know what is the best way to pull the latest libraries from front end?

makermelissa, 2024-02-24:
> curious to know what is the best way to pull the latest libraries from front end?

Here's one way I've done it before: https://github.com/adafruit/Adafruit_Dynamic_Bundler

s-light, 2024-03-25:
i like the idea to program CP in a visual way 

just some thoughts on this 
- get support for CP in [OpenRoberta lab](https://lab.open-roberta.org/) ([git](https://github.com/OpenRoberta/openroberta-lab)) 
- add only the [blocky](https://github.com/OpenRoberta/blockly) editor to this `web-editor`
- my favorite: use a node based visual flow based editor similar to http://noflojs.org/

i personally like the flow based version more than the blocky approach as its more of a real visual appealing design..

in general i think it would need a automagic system to *parse* the libraries and create *components* for them.
(maybe as a build job on the library builds...)
as otherwise you have no libs to use.. (its a to big burden for the lib maintainers to generate something special for the editor i think...)

what do you think about this?

would be a PR including some simple beginner friendly visual editor be an option?
i think it would make most sens to include this in the web-editor...
this way the visual editor is just a different representation of the normal code.
(and with this written down i think it will be difficult to follow the flow based approach - as it will be ?very difficult? to generate a flow out of some hand written code?!)

urfdvw, 2024-03-25:
> i like the idea to program CP in a visual way
>
> just some thoughts on this
>
> * get support for CP in [OpenRoberta lab](https://lab.open-roberta.org/) [...]

Hi @s-light !

Have you checked https://make.playpiper.com/ before?
It provides a block based CPY editor and packages for components.
There you can ignore the python part, or open the tab to see the generated python code from the blocks.
I am not sure if you can edit some part of the python code directly.

s-light, 2024-03-25:
no i did not tried this..
as fare as i can tell it uses the blocky editor / MS make.
where did you find the *packages* ?
do you know if the editor implementation from them is opensourced?

just also found 
https://think.cs.vt.edu/blockpy/blockpy/load

so to integrate blockly  should be possible..

s-light, 2024-03-26:
i extracted this [in its own featurerequest](https://github.com/circuitpython/web-editor/issues/171)

ericzundel, 2025-07-11:
Hi there! I'm super excited about this project!

As a developer, my top requests would be:
- open mutliple buffers/files in separate tabs
- An automatic python code formatter (like pep8 or black)

For my students, I'd like to see
- Renaming/backing up an existing file (they are constantly overwriting code.py)
- Easily uploading of libraries into the lib directory.  Most frequently they are uploading something from Adafruit CircuitPython Bundle.
- Download and install the latest CircuitPython onto the microprocessor from the UI with some instructions.
- Reconnecting to the microprocessor if the page is reloaded (this is fairly easy

More detailed notes in this [publicly shared Google Doc](https://docs.google.com/document/d/15LzXQ4xb5lAgQI85pmcsJGKNvTaukJxCVWa3gWRb5CA/edit?tab=t.0)

makermelissa, 2025-09-05:
I'm going to start breaking the most of these items out into enhancement issues and close this when done.

makermelissa, 2025-09-09:
Ok, I created issues for most of the ideas here. Any additional ideas should be added to https://github.com/circuitpython/web-editor/discussions/378.

closed 2025-09-09 by makermelissa