circuitpython/web-editor #21: Disconnecting in USB workflow cause the page to freeze.

urfdvw, 2022-10-19:
Test condition:
- Windows 10 with Chrome 106.0.5249.119 
- Seeed RP2040
- CircuitPython 7.3.3

Procedure:
- Connect the microcontroller with USB workflow
    - without opening any file
    - from browser console, seeing the serial output
        - which confirms that the code is running
- Click on the `Disconnect` button on the top-right corner
- Result: the browser tab freezes.

Tried:
- checked browser console, seeing log that the same info if repeating fast.
    - probably there is a dead loop caused by disconnection.

[image]

makermelissa, 2022-10-19:
Yeah, USB workflow is the newest workflow I added and it definitely needs some work. Thanks for creating the issue.

urfdvw, 2022-10-19:
Personally, I feel like Disconnecting Serial is not a first class function.
I don't think that is necessary for USB workflow,
as most of the time, closing the browser tab will disconnect,
and there is not a good reason for me to disconnect in the middle of working on a project.

Correct me if I miss anything.

makermelissa, 2022-10-19:
If you only have a single device, that's fine. However, is if you would like to connect serial to a different usb device or switch workflows then disconnecting is needed.

urfdvw, 2022-10-19:
> If you only have a single device, that's fine. However, is if you would like to connect serial to a different usb device or switch workflows then disconnecting is needed.

OK I get the switch workflow part. That is interesting.

makermelissa, 2022-12-02:
As I've been working on redoing USB workflow, I think I've found the cause of the issue. There's a break, but it's only breaking out of a single loop. So what happens is the read loop stops and then it goes into some kind of panic where it continually starts and stops the read loop causing it to freeze.

PR #42 (2022-12-05): USB Workflow Initial Rewrite
This is a bit rough around the edges, but functional.

Fixes #15. Fixes #21.

Oh yeah, it also fixes an issue with xterm not resizing properly sometimes.
closed 2022-12-05 by tannewt