circuitpython/web-editor #37: Use web workflow port to connect to the board

Neradoc, 2022-11-15:
The web workflow doesn't take into account the board's port, and always tries to connect to 80.
When the board is configured uses a different port, it can't connect back to it. (`CIRCUITPY_WEB_API_PORT`)

> Unable to connect. The device 192.168.1.92 was not found. Be sure it is plugged in and set up properly.

makermelissa, 2022-11-15:
I was wondering about that, hence #29.

Neradoc, 2022-11-15:
I swear I looked at the issues and missed it somehow 🤦 

I'm not setup to test locally, would changing from `location.hostname` to `location.host` here do it ?

https://github.com/circuitpython/web-editor/blob/7537516c728f6aca2ac4196768f26450ab5fbfd5/assets/js/workflows/web.js#L105-L109

makermelissa, 2022-11-16:
I was thinking of just setting up a couple of devices and changing the port to something besides 80 and seeing what happens.

makermelissa, 2022-11-18:
The error message that the person is running into is thrown here: https://github.com/circuitpython/web-editor/blob/main/assets/js/workflows/web.js#L146. There are likely several places that need fixing for alternate ports to work.

closed 2022-11-22 by tannewt

makermelissa, 2022-11-22:
> I'm not setup to test locally, would changing from `location.hostname` to `location.host` here do it ?

Yep, that's basically the fix. For some reason I was thinking there was extra stuff in host (like the protocol), but after looking it up just now, that would have been better than my solution.

PR #39 (2022-11-22): Fix non-standard port usage and Run Code import bugs
Fixes #37. Fixes #38.