circuitpython/web-editor #40: code.circuitpython.org: navigator.bluetooth.getDevices is not a function

makermelissa, 2022-11-23:
### CircuitPython version

```python
adafruit-circuitpython-microbit_v2-en_US-7.3.3.combined.hex; microbit_v2
```

### Code/REPL

```python
code.circuitpython.org
```

### Behavior

```
Uncaught (in promise) TypeError: navigator.bluetooth.getDevices is not a function

Argh: NetworkError: Connection failed for unknown reason.
```

### Description

[image]

beaufortfrancois, 2022-11-24:
`navigator.bluetooth.getDevices()` is available only when the user enabled the flag "chrome://flags/#enable-web-bluetooth-new-permissions-backend". You can follow https://bugs.chromium.org/p/chromium/issues/detail?id=577953 progress to know when this feature will be enabled by default for all users.

makermelissa, 2022-11-24:
That's correct, but there's code in place that should have detected and alerted the user. So a part of the issue is figuring out why that didn't happen.

beaufortfrancois, 2022-11-25:
I believe it's because there's no `try/catch` around these calls below:
https://github.com/circuitpython/web-editor/blob/57ddcbb4f5860ea345a3b17bc6ab2b4bd5a2a5b6/assets/js/workflows/ble.js#L66

https://github.com/circuitpython/web-editor/blob/57ddcbb4f5860ea345a3b17bc6ab2b4bd5a2a5b6/assets/js/workflows/ble.js#L242

This call below on the other hand is properly handled.

https://github.com/circuitpython/web-editor/blob/57ddcbb4f5860ea345a3b17bc6ab2b4bd5a2a5b6/assets/js/workflows/ble.js#L101 is caught

makermelissa, 2022-11-25:
Yeah, it's probably a good idea to add that.