circuitpython/web-editor #45: Redirect automatically from HTTP to HTTPS

spovlot, 2022-12-17:
If you just type code.circuitpython.org into the browser, then the USB connection will not work.  You must be on a secure site (https://code.circuitpython.org/) for the JavaScript to work correctly. 

The website should automatically redirect to HTTPS if it goes to the HTTP site.

spovlot, 2022-12-17:
This can be done via a Pages setting in the repository.  See https://docs.github.com/en/pages/getting-started-with-github-pages/securing-your-github-pages-site-with-https .

makermelissa, 2022-12-17:
Good call. I think @tannewt turned that off when we weren't sure how the web workflow and CORS stuff was going to work out. However, now that it's all sorted out, it should be fine to turn back on.

tannewt, 2022-12-19:
Checked the box!

closed 2022-12-19 by tannewt