circuitpython/web-editor #50: Add newline after entries in list of discovered devices.

dgriswo, 2023-01-30:
For ease of reading, it would be useful if a newline was added between device entries.
 
Screenshot of multiple boards attached.

[image]

makermelissa, 2023-01-30:
Thanks for the suggestion. Yeah, I agree it could use a bit more padding (from left and between items) to separate the items, which should make it easier to read. I'll just use css rather than using an entire new line though as this is meant to also work with mobile devices and space is limited.

makermelissa, 2023-02-07:
Ok, after playing around with this, now I get what you mean. Yeah, it should be 1 device per line. I think the display css property just needs to be updated or something simple like that.

makermelissa, 2023-02-07:
Fixed via #55

closed 2023-02-07 by makermelissa