circuitpython/web-editor #60: `TypeError: self._verifyPermission is not a function` when connecting over USB

huntercaron, 2023-02-11:
I did not see a bug template or anything sorry, so I will try to give all the info I have here.

## Bug

When trying to use the web editor I am running into the following error:

[image]

I don't see `self` being referenced anywhere else in the codebase, should it be a `this`?

## Info 
```
Adafruit CircuitPython 8.0.0 on 2023-02-06; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
UID:E66141040356482F
MAC:28:CD:C1:04:37:4B
```

makermelissa, 2023-02-13:
Yes, good catch. It's probably due to too much switching back and forth between python and javascript. :) Especially since I was working on a python/javascript interface at the time.

makermelissa, 2023-02-13:
Also, good idea on adding issue templates.

PR #62 (2023-02-13): Minor Bug Fixes to USB File Transfer
Fixes #60. This also fixes a misnamed reference to a function.
closed 2023-02-13 by tannewt