circuitpython/web-editor #64: Stuck on loading animation when saving file over USB

Neradoc, 2023-02-25:
On MacOS, with a RP2040 Zero board.

- connected to the board with the USB workflow
- allowed access to the drive (not called CIRCUITPY)
- write in a new file `print("Hello")`
- save as "text.py" in the board's drive via the Save as button
- page stuck on the rotating "loading blinka" animation

The following message output repeats in the console:
```js
write
usb-file-transfer.js:150 Setting modification time not currently supported in USB Workflow.
writeFile @ usb-file-transfer.js:150
await in writeFile (asynchrone)
writeFile @ file.js:67
writeFile @ workflow.js:285
up @ script.js:393
saveFile @ workflow.js:232
saveFileAs @ workflow.js:249
await in saveFileAs (asynchrone)
(anonyme) @ script.js:77
script.js:400 write failed ReferenceError: encoder is not defined
    at oa.writeFile (usb-file-transfer.js:156:13)
    at async Nl.writeFile (file.js:67:13)
    at async aa.showBusy (workflow.js:149:22)
    at async aa.writeFile (workflow.js:284:16)
    at async aa.up [as _saveFileContents] (script.js:393:13)
    at async aa.saveFile (workflow.js:232:13)
    at async aa.saveFileAs (workflow.js:249:17)
    at async HTMLButtonElement.<anonymous> (script.js:77:20) ReferenceError: encoder is not defined
    at oa.writeFile (https://code.circuitpython.org/assets/js/script.js:13:46341)
    at async Nl.writeFile (https://code.circuitpython.org/assets/js/script.js:13:3175)
    at async aa.showBusy (https://code.circuitpython.org/assets/js/script.js:13:30340)
    at async aa.writeFile (https://code.circuitpython.org/assets/js/script.js:13:32214)
    at async aa.up [as _saveFileContents] (https://code.circuitpython.org/assets/js/script.js:16:769)
    at async aa.saveFile (https://code.circuitpython.org/assets/js/script.js:13:31429)
    at async aa.saveFileAs (https://code.circuitpython.org/assets/js/script.js:13:31685)
    at async HTMLButtonElement.<anonymous> (https://code.circuitpython.org/assets/js/script.js:15:1877)
script.js:79 Current File Changed to: null
```

makermelissa, 2023-02-27:
Thanks, this should be a quick bug fix. It looks like it's missing the `let` keyword on https://github.com/circuitpython/web-editor/blob/main/assets/js/common/usb-file-transfer.js#L156.

makermelissa, 2023-02-27:
Oops, no. It's defined in the function header. I'll take a look though. I like these little puzzles. :)

PR #65 (2023-02-27): Fix encoder not defined error
Fixes #64. I also commented out the file time warning because it's not really important and could confuse users.
closed 2023-02-27 by tannewt