circuitpython/web-editor #84: Editor styling/formatting of single-line selections broken

Kriechi, 2023-04-06:
### CircuitPython Version

```python
irrelevant
```

### Workflow(s)

BLE, USB, WEB

### Browser(s)

```python
Version 111.0.5563.146 (Official Build) (x86_64)
```

### Steps

Click and drag to select a single word or multiple words in a single line.
Notice that the background color of the selected words does not change.
Once the selection spans multiple lines, a bright orange selection background appears.

### Additional information

https://user-images.githubusercontent.com/114300/230450196-40c72ca6-a4bd-4550-8856-5e8384ba6e88.mov

makermelissa, 2023-04-11:
This could be related to #79.

Kriechi, 2023-04-11:
Without having tested or bisect'd this: how would a terminal xterm change cause the editor styling to get messed up?

makermelissa, 2023-05-24:
I think this may be a a css issue. When I comment out the editor css file, it's difficult to read, but selection works as expected. Now to narrow down the culprit...

PR #101 (2023-05-25): Remove breaking CSS
Fixes #84. Fixes #97.

Removed some unused css as well as breaking css.
closed 2023-05-25 by tannewt