circuitpython/web-editor #90: https://github.com/adafruit/circuitpython/issues/7929

casainho, 2023-05-03:
### CircuitPython Version

```python
See: https://github.com/adafruit/circuitpython/issues/7929
```

### Workflow(s)

WEB

### Browser(s)

```python
See: https://github.com/adafruit/circuitpython/issues/7929

Firefox and Chrome, Linux, Android and MAC.
```

### Steps

See: https://github.com/adafruit/circuitpython/issues/7929

### Description

See: https://github.com/adafruit/circuitpython/issues/7929

makermelissa, 2023-05-03:
This was likely caused by #77. I'll take a look at it.

makermelissa, 2023-05-03:
Ok, I was able to reproduce the issue. Coincidentally, the only board I was able to find happened to be the same one you have. I'm almost certain I was correct about the cause of the issue now.

makermelissa, 2023-05-03:
Here's the console output:

```
index.css:1     Failed to load resource: the server responded with a status of 404 (Not Found)
loading-blinka.webp:1     Failed to load resource: the server responded with a status of 404 (Not Found)
index.js:1     Failed to load resource: the server responded with a status of 404 (Not Found)
index.css:1     Failed to load resource: the server responded with a status of 404 (Not Found)
```

makermelissa, 2023-05-03:
Fixed by #91.

closed 2023-05-03 by makermelissa

casainho, 2023-05-03:
Works as expected, thank you!! :+1: :partying_face: 

[image]

[image]

[image]