circuitpython/web-editor #102: Can not access from NAT (ngrok)

wuhao19920619, 2023-05-25:
### CircuitPython Version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; ESP32-S3-Box-Lite with ESP32S3
```

### Workflow(s)

WEB

### Browser(s)

```python
Google Chrome 113.0.5672.127
```

### Steps

1, Install Ngrok/oray or any NAT app on the router.
2, Forward Circuitpython device port `80` to the public internet port `8333` .
3, Access NAT domin (ex: `84c5df439d74.ngrok-free.dev:8333`) from the public internet.
4, After I see the Welcome page, click the `full code editor`.

### Description

In the code editor, it shows my devices can not be connected. But both welcome page and file browser work fine. I guess maybe the reason is `repl` in `full code editor` use another port to connect? Most of free NAT support only 1 port access.

wuhao19920619, 2023-05-27:
Well actually I can just use http. Currently the Welcome page and file browser works.

Don't need to worry about the https things. Because the services provider such as Oray will automaticlly remap the NAT domin to your local machine.

makermelissa, 2025-07-03:
Oh, yeah. I think the devices only work with http because it's simpler. As i mentioned above, I don't think its currently possible. If the devices handle https at some point, I think this can be remedied.

closed 2025-07-03 by makermelissa