circuitpython/web-editor #103: Serial and Editor butto out of the page

wuhao19920619, 2023-05-28:
### CircuitPython Version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; ESP32-S3-Box-Lite with ESP32S3
```

### Workflow(s)

WEB

### Browser(s)

```python
Google Chrome 113.0.5672.127
```

### Steps

[image]
If the code is longer than 1 single page, the Serial and Editor button will dissappear. So I have to reopen something short in order to open the serial monitor.

closed 2023-05-31 by makermelissa

Kriechi, 2023-05-31:
Looks like I'm a bit too late, but I believe the fix to this would be rather straight forward:

```
──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
modified: sass/layout/_editor.scss
──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
@ sass/layout/_editor.scss:8 @
     line-height: 1.5;
     font-family: 'Operator Mono', 'Source Code Pro', Menlo, Monaco, Consolas, Courier New, monospace;

+    max-height: calc(100vh - 2*5em - 4em);

     .cm-content {
         caret-color: orange;
     }
──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
modified: sass/layout/_layout.scss
──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
@ sass/layout/_layout.scss:7 @
     flex-direction: column;

     header {
-        // height: 5em;
+        height: 5em;
     }

     #footer-bar {
-        // height: 5em;
+        height: 5em;
         padding: 0 10px;
         display: flex;

@ sass/layout/_layout.scss:64 @
     flex-wrap: wrap;
     align-items: center;
     padding: 0 10px;
-    min-height: 60px
+    height: 4em;
 }
```

makermelissa, 2023-06-01:
@Kriechi, I can try that out tomorrow and if it works, I can revert the revert. :)

makermelissa, 2023-06-01:
I tested it out. It's close, but needs some adjusting because there's a gap that gets larger as the window gets smaller. I'd also like to test on mobile devices as well. The previous code did some adjusting for mobile because it wasn't taking into account the extra space for the navigation at the bottom, but that was removed by #86.

makermelissa, 2023-06-01:
Removing the `height: 100%` from `#main-content` fixed the gap issue.

makermelissa, 2023-06-07:
@Kriechi, I have re-merged the code along with necessary fixes for the gap and mobile display issues.

PR #106 (2023-05-31): Revert #86 side-by-side layout for editor+serial
Fixes #103.