circuitpython/web-editor #104: Freeze on the animation loading even when the terminal is working.

wuhao19920619, 2023-05-28:
### CircuitPython Version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; ESP32-S3-Box-Lite with ESP32S3
```

### Workflow(s)

USB, WEB

### Browser(s)

```python
Google Chrome 113.0.5672.127
```

### Steps

It occurs occasionally and I am not sure how to reproduce it. I have uploaded a video for this, you can see that the terminal is still working but the page freeze.

[https://youtu.be/I5SIlLrzduc](url)

When the page is loading, the loading animation prevents you from doing anything in your code. So if the device connection fails for any reason, the page will just freeze on the loading animation notification. There is even no chance to copy and paste your code, so everything unsaved will be lost. This can be improved and is necessary, I think.

makermelissa, 2023-05-31:
How long has this been happening? I'm wondering if this was caused by #86 which was merged in 3 weeks ago. This code has now been reverted by #106 and I'm hoping that fixed this issue as well.

wuhao19920619, 2023-05-31:
So far the issue still exists. I think the most urgent one is to modify the animation loading process. So at least it gives a chance to save code when the issue occurs.

Kriechi, 2023-05-31:
@wuhao19920619 if the issue occurs again, please open the browser dev tools and copy the console log message. Maybe they contain a hint of what is happening. Please paste them here as new comment.

PR #109 (2023-06-12): Runs with latest repl lib + fixed loading image
Fixes #104. Fixes #99. 

The following issues are resolved with this PR:
1. Fixes were made in the REPL library to work with this, so it now points to the latest version. 
2. The device.js was still slightly broken causing the loading animation to not appear correctly.