circuitpython/web-editor #118: Highlight text issue has resurfaced

makermelissa, 2023-06-23:
### CircuitPython Version

```python
None
```

### Workflow(s)

BLE, USB, WEB

### Browser(s)

```python
Google Chrome 114.0.5735.133
```

### Steps

1. Type text into editor
2. Select text
3. Notice selected text is not highlighted

pumatech, 2023-10-27:
Love this product and would use it in my high school classroom if not for this bug!

I think I have tracked down its origin - it's not that the selected-text isn't being highlighted, it's that there isn't sufficient contrast between the background color and the highlight color.  The default selected-text color in the Code Mirror dark theme is `#222` and the background color for class `.cm-editor` is `#333` (which is set by the application and is not part of the default Code Mirror dark theming, I believe.  I think it is defined in sass/layout/_editor.scss).  

This provides almost no contract between the background and the selected text.  If you open DevTools in Chrome and chase down the div with class `.cm-editor` you can change the background color to something lighter like `#444` and the highlight becomes visible.

Thanks for developing this code - it has the potential to make a huge difference for my students who only have Chromebooks to develop CircuitPython on for my classes.

jwfuller, 2023-11-01:
Looks like this change is still being overridden by the selector `.ͼ3 .cm-selectionBackground`. I will submit another PR.

Do you want me to revert the first change?

makermelissa, 2023-11-01:
> Looks like this change is still being overridden by the selector `.ͼ3 .cm-selectionBackground`. I will submit another PR.
> 
> Do you want me to revert the first change?

Nah, you can just include the revert in the next PR.

PR #145 (2023-10-31): Change background color for selected text
I think this resolves #118 but a more specific selector for `.c3 .cm-selectionBackgroud` may be needed. Since `.c3` isn't in the local SASS yet, I wanted to start with this change.
PR #146 (2023-11-01): Selector background v2
Revert 4bef14191707d11d42cd3a7f0561a990bb057da4 and try with a more specific selector. Tested by hand in a browser, not able to build to project at the moment. Will hopefully resolve #118