circuitpython/web-editor #189: Web Workflow Serial screen on Full Editor page is not accepting keyboard input

RetiredWizard, 2024-05-14:
### CircuitPython Version

```python
Adafruit CircuitPython 9.0.0-beta.2 on 2024-02-20; Adafruit QT Py ESP32C3 with ESP32-C3FN4
Adafruit CircuitPython 9.1.0-beta.1-19-g98988a1849 on 2024-05-10; Ai Thinker ESP32-CAM with ESP32
```

### Workflow(s)

WEB

### Browser(s)

```python
Google Chrome Version 124.0.6367.201 (Official Build) (64-bit)
```

### Steps

1. Connect to board via Web Workflow
2. Select Full Code Editor
3. Close the two dialog messages
4. Click on the serial button
5. Click within the serial window
6. Attempt to enter text, press return or ctrl-c/ctrl-d

### Description

Nothing happens in the serial window, it doesn't respond to any keyboard input. The Restart and Clear buttons both function properly.

makermelissa, 2024-05-14:
I wonder if #187 introduced this issue. I'll do some testing today.

closed 2024-05-14 by tannewt

RetiredWizard, 2024-05-14:
Thanks! Seems to work fine now :grin:

PR #190 (2024-05-14): Serial Console Bug fix
Fixes #189. Formatting on the function did not need to be anonymous, which was causing issues.