circuitpython/web-editor #192: Seeing two copies of the header at a certain breakpoint

tyeth, 2024-05-24:
### CircuitPython Version

```python
I was messing around with hijacking the websockets to recreate the bluefruit connect packets inside http://IP/code/ and spotted there were two connect buttons, but then I actually saw both at once, seemingly picking the perfect resolution to get both small and big nav menus.
```

### Workflow(s)

WEB

### Browser(s)

```python
windows 11, chrome 125.0.6422.76 (Official Build) (64-bit)
```

### Steps

Move window to half screen

### Additional information

[image]

tyeth, 2024-05-24:
same happens at code.circuitpython.org 

[image]

makermelissa, 2024-05-24:
Which browser are you using? It looks like it maybe allows fractional windows sizes between 767px and 768px, which might be causing it. This problem doesn't seem to be reproducible on Chrome for MacOS.

makermelissa, 2024-05-24:
Never mind, I see you already answered my question, I just missed it the first time.

makermelissa, 2024-05-24:
It might be caused by a discrepency between using min and max here:
https://github.com/circuitpython/web-editor/blob/main/sass/layout/_header_mobile.scss#L99
and here:
https://github.com/circuitpython/web-editor/blob/main/sass/layout/_header.scss#L138

makermelissa, 2024-05-24:
Ok, looks like I was correct about the fractional window sizing. I just noticed this in your screenshot:
[image]

Which means, I was probably correct about the use of both min and max. Ideally, they should be the same.