circuitpython/web-editor #202: Cant open/edit html files

aguaviva, 2024-06-25:
### CircuitPython Version

```python
Adafruit CircuitPython 9.1.0-beta.3 on 2024-05-22; Ai Thinker ESP32-CAM with ESP32
Board ID:ai-thinker-esp32-cam
```

### Workflow(s)

WEB

### Browser(s)

```python
Not dependent on the browser
```

### Steps

create a file as html, then try to open it to edit it.

### Description

[image]

makermelissa, 2024-07-01:
I think it's just a matter of adding to a list of text files.

aguaviva, 2024-07-01:
Is this something you want to take on?

makermelissa, 2024-07-02:
Yeah, it should be pretty easy. However, syntax highlighting will be a tougher beast to tackle in the future as it is currently python only.

PR #206 (2024-07-03): Add additional file types including html
Fixes #202. Also alphabetized the file types.
closed 2024-07-03 by tannewt