circuitpython/web-editor #203: Matrix Portal S3 & Feather RP2040 Scorpio - CIRCUITPY drive will not connect to code.circuitpython.org via USB.

Speedy-VI, 2024-06-28:
### CircuitPython Version

```python
Adafruit CircuitPython v9.0.5, Matrix Portal S3, Feather RP2040 Scorpio - all libraries updated with Circup.
```

### Workflow(s)

USB

### Browser(s)

```python
Google Chrome Version 126.0.6478.127  (Official Build) (64-bit)
```

### Console Log

```python
VM34:1 Uncaught 
SyntaxError: Unexpected identifier 'API'
    at neweggagent.js:1:10265
    at new Promise (<anonymous>)
    at Object.loadPageScript (neweggagent.js:1:10075)
    at neweggagent.js:1:9228
    at u (neweggagent.js:1:2927)
    at Generator._invoke (neweggagent.js:1:2680)
    at Generator.next (neweggagent.js:1:3286)
    at n (neweggagent.js:1:999)
    at s (neweggagent.js:1:1210)
    at neweggagent.js:1:1269

code.circuitpython.org/:1 Uncaught (in promise) 
Error: A listener indicated an asynchronous response by returning true, but the message channel closed before a response was received
code.circuitpython.org/:1 Uncaught (in promise) 
Error: A listener indicated an asynchronous response by returning true, but the message channel closed before a response was received
﻿
(Not sure if I did this correctly. I clicked on Errors in the console sidebar and copied the contents.)
```

### Steps

1 - open code.ciruitpython.org in the Chrome Browser
2 - Click on USB
3 - Click on Connect to Device
4 - In popup - Click on "CircuitPython CDC control (com26) - Paired"
5 - Click on Connect
6 - Popup window disappears, Spinning Purple Snake appears
7 - Nothing more happens

### Description

The following is for the Matrix Portal S3 but the Feather RP2040 Scorpio did the same thing. 

I opened code.ciruitpython.org in the Chrome Browser, clicked on USB, then Connect to Device, then in the popup window clicked on "CircuitPython CDC control (com26) - Paired", then Connect. When I do that, the popup window disappears and I get the spinning purple snake. It never prompted me to select the CIRCUITPY drive. It does something because the LEDs on the 64x32 RGB matrix I have attached to the Matrix Portal S3 turn off for a few seconds.

I posted about this on the Adafruit forum. Adafruit rep tannewt responded that this looked like a bug and asked me to report it here. There is more detail in the forum thread.
https://forums.adafruit.com/viewtopic.php?t=211722

makermelissa, 2024-07-01:
Thanks for letting us know. I'll take a look soon.

makermelissa, 2024-07-03:
I've been able to fairly consistently reproduce the issue. I believe the issue is in the code getting to a prompt. Essentially, it is sending the commands, but not waiting for the prompt which works in certain cases, but not always.

For instance, I had my ocde.py with the deprecated displayio show() command, which stopped executing and worked pretty consistently the way it is supposed to. Once I fixed the error in my code, the code was not properly interrupting the execution anymore and this problem appeared.

bablokb, 2024-07-04:
I suggest that you add an event-listener to the connect event of the Serial object and remove the spinner once it is connected (`usb.js`, function `_switchToDevice()`). This will not solve the issue for the code-editor, but at least for the serial console.

makermelissa, 2024-07-08:
> I suggest that you add an event-listener to the connect event of the Serial object and remove the spinner once it is connected (`usb.js`, function `_switchToDevice()`). This will not solve the issue for the code-editor, but at least for the serial console.

Yeah, it probably needs quite a bit of reworking to be more robust. So if there are errors or whatever, the whole thing doesn't hang.

makermelissa, 2024-07-08:
Ok, it's now connecting. There seems to be a long pause while connecting, but it does eventually connect. I'm still working on improving it and will leave this issue open until I'm happy with it. The speed should be much faster (like under 5 seconds at most).

Speedy-VI, 2024-07-09:
> Ok, it's now connecting. There seems to be a long pause while connecting, but it does eventually connect. I'm still working on improving it and will leave this issue open until I'm happy with it. The speed should be much faster (like under 5 seconds at most).

Thank you for the update. Should I try to connect again with my Matrix Portal S3 and Feather RP2040 Scorpio or wait until  you close the issue?

makermelissa, 2024-07-09:
> Thank you for the update. Should I try to connect again with my Matrix Portal S3 and Feather RP2040 Scorpio or wait until you close the issue?

You can connect now. However, you'll likely need to wait like 10-15 seconds for the dialog to update.

makermelissa, 2024-07-09:
Ok, I've determined part of the delay is because the REPL is sending a series of control characters in the REPL to try and get from whichever mode it is in to normal mode. Unfortunately, one of the characters it sends is CTRL+D to get from Raw Paste Mode to Raw Mode, but if the board is in Normal mode, it will CTRL+D will cause a restart, which delays things considerably. To get around this, the app needs to have the mode awareness improved and only send the appropriate control characters.

makermelissa, 2024-07-09:
I think I have it working reliably now, but want to do further testing before committing the code.

Speedy-VI, 2024-07-09:
> I think I have it working reliably now, but want to do further testing before committing the code.

Great news! In your last post, you said at least part of the problem was that the REPL was sending CNTRL+D in Normal mode. I think you are referring to the Mu editor and the app itself would need to be modified. Did you get around this somehow? I was going to try it but got a popup that says, "USB Workflow is currently experiencing issues. See GitHub issue #203 for more details. Please use Web Workflow.". I will wait until you commit the code. Thanks for your work on this.

makermelissa, 2024-07-09:
> Great news! In your last post, you said at least part of the problem was that the REPL was sending CNTRL+D in Normal mode. I think you are referring to the Mu editor and the app itself would need to [...]

You're welcome. Mu is not involved at all. The Web Editor uses https://github.com/adafruit/circuitpython-repl-js which is a library I wrote to communicate with the REPL. Unfortunately there were some issues in the last major release and I am ironing them out. Testing has been going well. I was trying to work out some minor things such as content printing to the REPL that isn't supposed to be, but I may just release it to get USB workflow working properly.

closed 2024-07-09 by dhalbert

makermelissa, 2024-07-09:
Code is updated and message has been removed. You are free to try it out.

Speedy-VI, 2024-07-09:
> Code is updated and message has been removed. You are free to try it out.

I just tried it with my Matrix Portal S3. It's working! Thanks again.

bablokb, 2024-07-10:
I still have problems. The editor seems to connect in the background, but the connect popup is still visible. Good thing: I can now close it. After closing I can see in the serial tab the raw REPL. My device has an integrated display and I can also see the REPL on the display. 

Using some code in the editor-tab and pressing "save+run" shows no reaction. Pressing the connect button in the header on the right will print "disconnected" in the raw REPL. Pressing connect again will bring back the raw REPL.
[image]

So maybe you reopen this issue again?!

makermelissa, 2024-07-10:
Yeah, I'm going to rework it to use Raw mode instead of Raw Paste mode, which I think will fix some weird issues like that which we've noticed as well.

PR #212 (2024-07-09): Update to latest REPL lib to fix delays
Fixes #203. Tested for quite a while and it has been pretty reliable. This also removes the message about the USB workflow issue.