circuitpython/web-editor #217: Create a Beta Test Site for the Code Editor

makermelissa, 2024-07-18:
In order to mitigate bugs in code from taking down the code editor, we should set up a beta site that we can push code to before pushing it to code.circuitpython.org. It should mirror the current code.circuitpython.org environment as closely as possible. This will give people who would like the latest features a chance to test the code out, but also a backup option with the main site if there is anything that stops the beta site from working.

I'm not sure what would be involved in setting this up, nor what the workflow would look like, so I'm open to input on that.

tannewt, 2024-07-18:
@jwcooper can make the subdomain. I don't know if github will host two domains from one repo.

dhalbert, 2024-07-18:
Some discussion of publishing multiple branches here: https://github.com/orgs/community/discussions/21582. This was hard to find.

Maybe the long-term solution is not to use github pages for this, if what's in the discussion above is not practical.

makermelissa, 2024-07-18:
Thanks for the info Dan. Multiple branches would be ideal. I think we should attempt that first before giving up on github.

makermelissa, 2024-07-19:
@jwcooper I'm leaning towards the CloudFlare approach. I have created a `beta` branch. I'm not quite sure how it would work as there is a little work done via GitHub actions which runs some npm commands. For the URL, were thinking of something like `code-beta.circuitpython.org`.

makermelissa, 2024-07-22:
@jwcooper if you can't get it to work, I may just make a new repo to keep things simple for now.

jwcooper, 2024-07-22:
I haven't given it a go yet. If we want to try, I'll need access to this repo/org. Cloudflare pages connects via github auth.

Honestly though, a forked repo may be the simplest approach too.

makermelissa, 2024-07-22:
> Honestly though, a forked repo may be the simplest approach too.

Yes, except I don't believe GitHub allows forks on the same account (I tried with the WebSerial ESPTool). Maybe we can have it reside on the main Adafruit Account. @tannewt what are your thoughts on that?

makermelissa, 2024-07-22:
> I haven't given it a go yet. If we want to try, I'll need access to this repo/org. Cloudflare pages connects via github auth.

Ok, I added you as a maintainer to this repo.

jwcooper, 2024-07-22:
I requested cloudflare pages approval.

dhalbert, 2024-07-22:
I approved that a few minutes ago but I of course could not log in to cloudflare to finish the job. Maybe you already did it for yourself.

jwcooper, 2024-07-22:
@makermelissa Once dns propagates, this page should load: https://code-beta.circuitpython.org/

Cloudflare pages will build/deploy on every commit to the `beta` branch.

@dhalbert thank you for your help as well.

makermelissa, 2024-07-22:
Thank you @jwcooper and @dhalbert.

makermelissa, 2024-07-22:
It's working for me.

makermelissa, 2024-08-05:
Ok, it doesn't look like when code gets merged into the beta branch that it's going live on code-beta.circuitpython.org.

jwcooper, 2024-08-05:
It's failing on build:

```
16:17:04.865 | Error: Cannot find module @rollup/rollup-linux-x64-gnu. npm has a bug related to optional dependencies (https://github.com/npm/cli/issues/4828). Please try `npm i` again after removing both package-lock.json and node_modules directory.
-- | --
```

makermelissa, 2024-08-05:
@jwcooper I was running into that and updated the actions file. It is passing now, however, when going to code-beta.circuitpython.org, it is not showing new changes. Is is possible that there's a long cache time?

jwcooper, 2024-08-05:
@makermelissa cloudflare isn't using the actions file, it's using the package.json/package-lock.json directly via `npm run build` to execute vite.

Here is the full build log:

```
2024-08-05T21:53:15.021885Z	Cloning repository...
2024-08-05T21:53:15.679891Z	From https://github.com/circuitpython/web-editor
2024-08-05T21:53:15.680644Z	 * branch            882f95603dd8dc00d2c5e5d3a2420345c8e6a3ea -> FETCH_HEAD
2024-08-05T21:53:15.680803Z	
2024-08-05T21:53:15.716976Z	HEAD is now at 882f956 Merge pull request #225 from makermelissa/beta
2024-08-05T21:53:15.717409Z	
2024-08-05T21:53:15.800228Z	
2024-08-05T21:53:15.800814Z	Using v2 root directory strategy
2024-08-05T21:53:15.826692Z	Success: Finished cloning repository files
2024-08-05T21:53:17.496423Z	Checking for configuration in a wrangler.toml configuration file (BETA)
2024-08-05T21:53:17.496996Z	
2024-08-05T21:53:17.597239Z	No wrangler.toml file found. Continuing.
2024-08-05T21:53:17.744863Z	Detected the following tools from environment: npm@9.6.7, nodejs@18.17.1
2024-08-05T21:53:17.745554Z	Installing project dependencies: npm clean-install --progress=false
2024-08-05T21:53:22.046382Z	
2024-08-05T21:53:22.046634Z	added 63 packages, and audited 64 packages in 4s
2024-08-05T21:53:22.046811Z	
2024-08-05T21:53:22.046895Z	6 packages are looking for funding
2024-08-05T21:53:22.046956Z	  run `npm fund` for details
2024-08-05T21:53:22.048495Z	
2024-08-05T21:53:22.048616Z	found 0 vulnerabilities
2024-08-05T21:53:22.061571Z	Executing user command: npm run build
2024-08-05T21:53:22.611569Z	
2024-08-05T21:53:22.611849Z	> web-editor@0.0.0 build
2024-08-05T21:53:22.611988Z	> vite build
2024-08-05T21:53:22.612091Z	
2024-08-05T21:53:22.775339Z	/opt/buildhome/repo/node_modules/rollup/dist/native.js:59
2024-08-05T21:53:22.775673Z			throw new Error(
2024-08-05T21:53:22.775885Z			      ^
2024-08-05T21:53:22.775974Z	
2024-08-05T21:53:22.776111Z	Error: Cannot find module @rollup/rollup-linux-x64-gnu. npm has a bug related to optional dependencies (https://github.com/npm/cli/issues/4828). Please try `npm i` again after removing both package-lock.json and node_modules directory.
2024-08-05T21:53:22.77632Z	    at requireWithFriendlyError (/opt/buildhome/repo/node_modules/rollup/dist/native.js:59:9)
2024-08-05T21:53:22.776963Z	    at Object.<anonymous> (/opt/buildhome/repo/node_modules/rollup/dist/native.js:68:76)
2024-08-05T21:53:22.777405Z	    ... 3 lines matching cause stack trace ...
2024-08-05T21:53:22.777514Z	    at Module._load (node:internal/modules/cjs/loader:960:12)
2024-08-05T21:53:22.777656Z	    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:169:29)
2024-08-05T21:53:22.777884Z	    at ModuleJob.run (node:internal/modules/esm/module_job:194:25) {
2024-08-05T21:53:22.777977Z	  [cause]: Error: Cannot find module '@rollup/rollup-linux-x64-gnu'
2024-08-05T21:53:22.778037Z	  Require stack:
2024-08-05T21:53:22.778097Z	  - /opt/buildhome/repo/node_modules/rollup/dist/native.js
2024-08-05T21:53:22.778754Z	      at Module._resolveFilename (node:internal/modules/cjs/loader:1077:15)
2024-08-05T21:53:22.77885Z	      at Module._load (node:internal/modules/cjs/loader:922:27)
2024-08-05T21:53:22.778974Z	      at Module.require (node:internal/modules/cjs/loader:1143:19)
2024-08-05T21:53:22.779584Z	      at require (node:internal/modules/cjs/helpers:121:18)
2024-08-05T21:53:22.779957Z	      at requireWithFriendlyError (/opt/buildhome/repo/node_modules/rollup/dist/native.js:41:10)
2024-08-05T21:53:22.780052Z	      at Object.<anonymous> (/opt/buildhome/repo/node_modules/rollup/dist/native.js:68:76)
2024-08-05T21:53:22.780519Z	      at Module._compile (node:internal/modules/cjs/loader:1256:14)
2024-08-05T21:53:22.780621Z	      at Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
2024-08-05T21:53:22.781027Z	      at Module.load (node:internal/modules/cjs/loader:1119:32)
2024-08-05T21:53:22.781114Z	      at Module._load (node:internal/modules/cjs/loader:960:12) {
2024-08-05T21:53:22.781572Z	    code: 'MODULE_NOT_FOUND',
2024-08-05T21:53:22.781675Z	    requireStack: [ '/opt/buildhome/repo/node_modules/rollup/dist/native.js' ]
2024-08-05T21:53:22.782183Z	  }
2024-08-05T21:53:22.782277Z	}
2024-08-05T21:53:22.782398Z	
2024-08-05T21:53:22.782604Z	Node.js v18.17.1
2024-08-05T21:53:22.790072Z	Failed: Error while executing user command. Exited with error code: 1
2024-08-05T21:53:22.800059Z	Failed: build command exited with code: 1
2024-08-05T21:53:23.842459Z	Failed: error occurred while running build command
```

makermelissa, 2024-08-05:
Oh, gotcha.

makermelissa, 2024-08-05:
I installed the new package and it has updated successfully.

closed 2024-08-05 by makermelissa