circuitpython/web-editor #229: writing code.py causes: OSError: [Errno 5] Input/output error

dhalbert, 2024-08-08:
### CircuitPython Version

```python
8.2.0
9.0.0
9.1.1
all of above tested on two RP2040 boards

also reproduced the problem on 9.1.1 on an ESP32-S3

did not reproduce on a Trinket M0
```

### Workflow(s)

USB

### Browser(s)

```python
Version 127.0.6533.99 (Official Build) (64-bit) on Ubuntu 24.04
```

### Console Log

```python
Load different workflow
index.js:91 Requesting any serial device...
index.js:91 > Requested  SerialPort {onconnect: null, ondisconnect: null, readable: null, writable: null}
index.js:91 switch to SerialPort {onconnect: null, ondisconnect: null, readable: null, writable: null}
index.js:93 Read Loop Init
index.js:93 Read Loop Started
index.js:91 Getting Device UID...
index.js:93 Device UID: F6254209F45633
index.js:91 Successfully loaded directory: CIRCUITPY
index.js:91 New folder name: CIRCUITPY
index.js:91 SerialPort {onconnect: null, ondisconnect: null, readable: ReadableStream, writable: WritableStream}
index.js:89 Connected!
index.js:89 Path: /code.py
index.js:93 Current File Changed to: /code.py

[no output when writing code.py]
```

### Steps

1. Shift reload https://code-beta.circuitpython.org
2. select USB and choose serial and CIRCUITPY for an RP2040 board (I've tried two). The board has previously had `storage.erase_filesystem()` done, so CIRCUITPY is in good shape.
3. open Serial window
4. edit `code.py` and another another `print()` statement. Note that saving code.py does NOT cause an auto-reload
5. Type ctrl-D in serial window. I get:
```
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
OSError: [Errno 5] Input/output error

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
The `OSError` is new. It did not show up if I did ctrl-D before editing the file.

### Additional information

**This does _not_ seem to reflect a corrupted CIRCUITPY.** Instead, it is some other I/O error, maybe due to something being done via serial. If I disconnect from web-editor and just connect via a terminal program, without resetting, and do a ctrl-D, I do not see the `OSError`.

makermelissa, 2024-08-09:
I wasn't able to exactly fix the issue, but did find a similar issue in the process and have slowly been working it out of the system. After that, I'll try the exact reproduction steps.

makermelissa, 2024-08-09:
Oh, I just realized it didn't happen on all boards. Will try on a RP2040 after I get the issue I notice worked out.

makermelissa, 2024-08-09:
Just tested on the Feather RP2040 running CP 9.1.1 on my system locally and did not experience the error, so perhaps it's fixed now. After I finish #230, I'll submit a PR.

makermelissa, 2024-08-10:
This should now be fixed via #231.

closed 2024-08-10 by makermelissa

FoamyGuy, 2024-08-12:
I'm able to re-create this issue reliably as well.

Browser: `Version 127.0.6533.99 (Official Build) (64-bit)`
OS: Ubunty 24.04

I've seen the same issue with a MEMENTO 9.1.1 and a Feather RP2040 9.0.4 and 9.1.1, it doesn't appear to be tied to a specific device or version that I can tell.

Here is a screen recording of one of occurrences on the MEMENTO
[Screencast from 2024-08-12 17-37-05.webm](https://github.com/user-attachments/assets/e50030fc-e656-4a0d-82b9-a40f3daf62e5)

I found that when the device is in the state where it's always getting the Input/Output error I can press the disconnect button at the top, and then re-connect and upon re-connecting open the serial and use ctrl-c / ctrl-d to run code.py and it does execute without the error as long as I do not change the code again before running. I have noticed what I think is an unrelated issue where disconnect / re-connecting to a device without reloading the page causes serial input and outputs to be duplicated, i'll create a seperate issue for that. 

Since it works after disconnect / reconnecting it seems to me that this issue is tied specifically to when it saves the code.py content and then tries to run it afterward. The actual content of the file seems to get saved correctly because it works without any further changes after the disconnect / reconnect.

makermelissa, 2024-08-13:
I wonder if the issue is only appearing on linux. I'm trying with Mac OS, but have a linux computer (running Linux Mint) I can try it on. I believe Dan was probably also trying on Linux.

dhalbert, 2024-08-13:
Yes, I only tested on Linux.

makermelissa, 2024-08-13:
First of all, it worked fine for me on linux. However, I may have an idea about what is happening. For the RP2040, there should be a CIRCUITPY drive that you need to grant permission to. If it is trying to go through the REPL (because maybe it detects the file system isn't read only), then I think that's why it is encountering the error.

Getting down a bit deeper, I remembered that in order to make it Micropython compatible, I have it attempting to check if the file system is read only and if there are any errors it assumes that it isn't read-only. I think I need to make it more specific to an import error.

makermelissa, 2024-08-14:
Ok, am trying the memento and noticing weirdness (code timeouts, slowness, not connecting with File System API, etc). Hopefully getting it working will iron out the issue. I think the weirdness is causing it to think it should do file ops through the REPL and then it can't write to a read-only file system.

makermelissa, 2024-08-14:
Ok, good news is same board works much better after trying on my local system (which only assumes not read-only on import error instead of all errors). I think I can improve the connection speed/weirdness though and can call this good.

makermelissa, 2024-08-14:
I just noticed in Dan's log:
```
index.js:91 New folder name: CIRCUITPY
```

This means that it is connecting through the File System API, which is correct. So, I don't think my hypothesis is correct.

makermelissa, 2025-02-12:
I removed the beta label because I don't think this has quite been fixed on beta yet.

makermelissa, 2025-02-14:
I think when trying to reproduce this before, I had missed the step where you typed Ctrl-D. Now after following the instructions above, I get:
```
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

This was on the Metro RP2350. This could actually be a CircuitPython bug. My suspicion is that maybe it does this after being put in a certain mode. I'll need to take a closer look at some of the underlying REPL module code.

makermelissa, 2025-02-14:
At some point, it does run some code to check if the filesystem is readonly. We may need to put it in normal mode if it is readonly and hopefully that will solve it from this end. However, if I can fiigure out how to reproduce the bug manually, this could also be fixed in CP.

makermelissa, 2025-02-14:
Ok, I couldn't reproduce by hand. Then I gave it another try and it worked as expected. I'll try this on an RP2040 feather next.

makermelissa, 2025-02-14:
I loaded CP 9.1.1 onto a RP2040 board and wasn't able to reproduce the issue. When one of you gets a chance, could you verify this is still an issue? I've made several bug fixes to circuitpython-repl-js and I'm hoping those fixed the bug. Please try with latest firmware as well in case this got fixed there.

makermelissa, 2025-06-30:
Since this issue came up more with linux computers, it makes sense to test this on a linux computer. If I am unable to reproduce the issue, we may need to close it until we find a reliable way to recreate the conditions of the error. I will give it a try using 9.1.1 to remove that as a variable. If I can reproduce the issue, then I will try on a more recent version of CP.

makermelissa, 2025-06-30:
I was finally able to reproduce the issue on linux and CP 9.1.1 on a Feather RP2040.

makermelissa, 2025-06-30:
The issue is still present in 9.2.8.

PR #231 (2024-08-10): Update REPL JS library to bug fixed version
Fixes #229. Fixes #230.