circuitpython/web-editor #230: Extraneous output in Serial window

dhalbert, 2024-08-08:
### CircuitPython Version

```python
CircuitPython 9.1.1 on QT Py ESP32-C3
```

### Workflow(s)

USB

### Browser(s)

```python
Version 127.0.6533.99 (Official Build) (64-bit) on Ubuntu 24.04
```

### Console Log

```python
nothing unusual in console log
```

### Steps

1. Shift reload https://code-beta.circuitpython.org/
2. select USB and choose serial. The board has previously had storage.erase_filesystem() done, so CIRCUITPY is in good shape.
3. open Serial window
4. edit code.py and another another print() statement

### Description

I see extraneous output in the Serial Window, copied below. I see soft restarts, raw REPL info, a few errors, the file I'm writing, etc.
```

raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKFalse
>

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
connected

>>> 
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKFalse
>

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKsd 16384 0 946684802
settings.toml 32768 0 946684802
code.py 32768 22 946684802
lib 16384 0 946684802
boot_out.txt 32768 135 946684804
>

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKprint("Hello World!")

>

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 6, in <module>
NameError: name 'os' is not defined
>

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!
foo

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!
foo

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKsd 16384 0 946684802
settings.toml 32768 0 946684802
code.py 32768 34 946684856
lib 16384 0 946684802
boot_out.txt 32768 135 946684804
>

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> 
raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKprint("Hello World!")
print("foo")
>
```

makermelissa, 2024-08-08:
Thanks. I'll look into this today.

makermelissa, 2024-08-09:
I noticed this as well and was able to reproduce the issue locally. I haven't traced down the exact cause yet.

makermelissa, 2024-08-10:
This is now via #231.

PR #231 (2024-08-10): Update REPL JS library to bug fixed version
Fixes #229. Fixes #230.
closed 2024-08-10 by makermelissa