circuitpython/web-editor #238: Issues with connecting to USB workflow

BlitzCityDIY, 2024-08-19:
### CircuitPython Version

```python
9.1, Adafruit Feather ESP32-C6
```

### Workflow(s)

USB

### Browser(s)

```python
Google Chrome Version 127.0.6533.120 (Official Build) (64-bit)
```

### Console Log

```python
Load different workflow
index.js:91 Requesting any serial device...
index.js:91 > Requested  SerialPort {onconnect: null, ondisconnect: null, readable: null, writable: null}
index.js:91 switch to SerialPort {onconnect: null, ondisconnect: null, readable: null, writable: null}
index.js:93 Read Loop Init
index.js:93 Read Loop Started
index.js:91 Requesting any serial device...
index.js:91 SerialPort {onconnect: null, ondisconnect: null, readable: ReadableStream, writable: WritableStream}
2index.js:91 Requesting any serial device...
```

### Steps

1. Reset board or power cycle (unplug/replug USB)
2. open new chrome window and navigate to code editor
3. Try to connect by selecting USB
4. The port shows up after clicking Connect Device. I select the port and click connect
5. The screen remains on the Select serial device window
6. If I open the console, it usually shows Read Loop Started where it freezes up

### Description

The initial connection I did worked basically instantly. After using it for a while and leaving it idle, the connection seemed to stop working. When I tried to reconnect, this bug started occurring. Sometimes I am able to get it to connect (screenshot below) but then if I try to open a file it hangs with the spinning Blinka.

### Additional information

[image]

makermelissa, 2024-08-19:
Thanks. I see there were 1393 Normal Prompt not Detected messages (which is where it's stuck). Have you verified that you can connect to the REPL using another tool?

BlitzCityDIY, 2024-08-19:
@makermelissa yes i can connect with putty:

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 16, in <module>
KeyboardInterrupt:

Code done running.
🐍Wi-Fi: off | BLE:CIRCUITPY597aae | REPL | 9.2.0-alpha.2350-18-g830294ebb5
Adafruit CircuitPython 9.2.0-alpha.2350-18-g830294ebb5 on 2024-08-16; Adafruit Feather ESP32-C6 4MB Flash No PSRAM with ESP32C6
>>>
>>>
>>>
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

```

makermelissa, 2024-08-19:
Ok, It's acting weird on mine. After a fresh install it works the first time and after reseting the device, it comes up "waiting for download" which is not what the REPL library is looking for.

makermelissa, 2024-08-19:
When you put CircuitPython on yours, which address did you install at?

BlitzCityDIY, 2024-08-19:
i used 0x0, i found the default 0x1000 didn't work and similarly on that first run, it connected right away but did not afterwards.

makermelissa, 2024-08-19:
> i used 0x0, i found the default 0x1000 didn't work and similarly on that first run, it connected right away but did not afterwards.

I suspect this is a CircuitPython issue. The Alpha doesn't even work at all.

BlitzCityDIY, 2024-08-19:
i was able to erase flash and load 9.1.1 and reestablish a connection. but initially this issue popped up when i was using 9.1.1 after i had been connected for an hour or two. i'll leave it connected to see if i can recreate it and get a printout from the console

BlitzCityDIY, 2024-08-19:
i was able to recreate the reconnection issue with 9.1.1. it occurs when editing settings.toml. i had the file opened in the editor. i saved my changes and then it hung for a bit. then the console filled with ctrl+c resets:

[image]

i couldn't access code.py or rerun code so i reset the board and refreshed the window. this caused a hang in reconnection:

[image]

it did eventually connect as you can see in the console, but i couldn't open code.py. i would just get the blinka spinning. on resetting/refreshing the window again i was able to reconnect after about 1-2 minutes. however, when i click open to access the files it hangs and the normal prompt error continuously counts up:

[image]

BlitzCityDIY, 2024-08-26:
i've done some additional testing and found two repeatable instances using 9.1.x and the non-beta editor:

1. if the board is running a code.py file that has a blocking delay in the loop, it will take a very long time to connect or will hang indefinitely depending on the length of the delay. i was able to test this with a simple blink example:

```
import time
import board
import digitalio

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

when i have the loop included, it will take a very long time to connect or fail. when i remove the loop, it connects right away as it does with a fresh CP install running `print("hello world!")`.

2. Editing and saving **settings.toml** always results in a crash in the editor and on the board

when i open and save settings.toml, the Serial window fills with board resets. When I try to open the file directory, it appears to be empty:

[image]

here is the text output from the Chrome console starting with opening settings.toml:
```
Path: /settings.toml
index.js:93 Current File Changed to: /settings.toml
index.js:74 Awaiting mode change timed out.
25index.js:74 Normal Prompt not detected.
index.js:74 Awaiting code interruption timed out. Restarting device.
15index.js:74 Normal Prompt not detected.
index.js:74 Awaiting mode change timed out.
67index.js:74 Normal Prompt not detected.
index.js:74 Code timed out.
_waitForCodeExecution @ index.js:74
await in _waitForCodeExecution
execRawMode @ index.js:74
await in execRawMode
runCode @ index.js:74
await in runCode
listDir @ index.js:53
listDir @ index.js:91
await in listDir
_openFolder @ index.js:91
open @ index.js:91
await in open
openFileDialog @ index.js:91
openFile @ index.js:91
await in openFile
Kp @ index.js:93
await in Kp
(anonymous) @ index.js:93Understand this error
54index.js:74 Normal Prompt not detected.
index.js:91 Path: null
73index.js:74 Normal Prompt not detected.
index.js:74 Awaiting code interruption timed out. Restarting device.
47index.js:74 Normal Prompt not detected.
index.js:74 Awaiting mode change timed out.
index.js:74 Unexpected output in raw mode: ]0;🐍Wi-Fi: off | BLE:CIRCUITPY597aae | Done | 9.1.2\
_checkCodeRunning @ index.js:74
await in _checkCodeRunning
(anonymous) @ index.js:74
await in (anonymous)
_timeout @ index.js:74
interruptCode @ index.js:74
getToPrompt @ index.js:74Understand this error
index.js:74 Unexpected output in raw mode: soft reboot
]0;����Wi-Fi: 201 | BLE:CIRCUITPY597aae | Done | 9.1.2\
_checkCodeRunning @ index.js:74
await in _checkCodeRunning
(anonymous) @ index.js:74
await in (anonymous)
_timeout @ index.js:74
_waitForCodeExecution @ index.js:74
execRawMode @ index.js:74
await in execRawMode
runCode @ index.js:74Understand this error
index.js:74 Unexpected output in raw mode: 

_checkCodeRunning @ index.js:74
await in _checkCodeRunning
(anonymous) @ index.js:74
await in (anonymous)
_timeout @ index.js:74
_waitForCodeExecution @ index.js:74
execRawMode @ index.js:74
await in execRawMode
runCode @ index.js:74
await in runCode
listDir @ index.js:53
listDir @ index.js:91
await in listDir
_openFolder @ index.js:91
open @ index.js:91
await in open
openFileDialog @ index.js:91
openFile @ index.js:91
await in openFile
Kp @ index.js:93
await in Kp
(anonymous) @ index.js:93Understand this error
index.js:74 Unexpected output in raw mode: ]0;🐍Wi-Fi: off | BLE:CIRCUITPY597aae | Done | 9.1.2\]0;🐍Wi-Fi: 201 | BLE:CIRCUITPY597aae | Done | 9.1.2\
_checkCodeRunning @ index.js:74
await in _checkCodeRunning
(anonymous) @ index.js:74
await in (anonymous)
_timeout @ index.js:74
_waitForCodeExecution @ index.js:74
execRawMode @ index.js:74Understand this error
index.js:74 Code timed out.
_waitForCodeExecution @ index.js:74
await in _waitForCodeExecution
execRawMode @ index.js:74
await in execRawMode
runCode @ index.js:74
await in runCode
listDir @ index.js:53
listDir @ index.js:91Understand this error
2index.js:74 Normal Prompt not detected.
```

after resetting and/or unplugging the board and refreshing chrome i can establish a connection but when i click Open to try and open code.py, the editor spins for a while. I eventually get the same result that I included in the screenshot, the Serial window filling with resets and an empty file directory.

I was able to connect to the board in this state with the Viper IDE. Initially it showed the board in the same state but then eventually refreshed to show the file system intact and what I had saved to settings.toml was there as well. After that I was able to connect again with code.circuitpython.org.

BlitzCityDIY, 2024-08-26:
i was testing with live. i just tried beta and i can't get it to connect at all. i erased flash and did a new install of 9.1.2. when i close out of the chrome window i see the board reset so there is some communication happening.

makermelissa, 2024-08-26:
Ok, thanks. That's what I'm seeing. I may have messed something up in my latest changes to the REPL lib.

makermelissa, 2024-08-26:
Ok, I rolled back a couple things on my system and it seems to be working in parity with the live version now.

makermelissa, 2024-08-26:
@BlitzCityDIY please try on code-beta again. I have fixed some bugs and tested myself and it appears to be working much better.

BlitzCityDIY, 2024-09-05:
@makermelissa i tried code-beta this morning. erased flash on the c6, loaded 9.1.3 and with the default hello world it connected right away. i updated code.py to be the built-in LED blink code. i refreshed the browser and disconnected/reconnected the C6 and the connection hung. when it did connect, the LED had stopped blinking so i opened the serial console:
```
>>> 
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Unable to allocate to the heap.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

this was the output in the Chrome console:

```
Load different workflow
index.js:94 Requesting any serial device...
index.js:94 > Requested  SerialPort
index.js:94 switch to SerialPort
index.js:96 Read Loop Init
index.js:96 Read Loop Started
index.js:92 Normal Prompt not detected.
index.js:92 Connected!
index.js:94 SerialPort {onconnect: null, ondisconnect: null, readable: ReadableStream, writable: WritableStream}
index.js:92 Path: /code.py
index.js:96 Current File Changed to: /code.py
```

should i file a CP issue citing this issue?

makermelissa, 2024-09-05:
> should i file a CP issue citing this issue?

Yeah, that makes sense.

tannewt, 2024-09-12:
I've got a CircuitPython-side fix for this. We weren't clearing the incoming serial buffer when receiving ctrl-c. This meant we skipped straight into the repl without printing "press any key". USB CDC and UART already did this clear.

closed 2024-09-13 by dhalbert