circuitpython/web-editor #240: Have a new dialog message appear on screen while board is connecting

BlitzCityDIY, 2024-08-26:
Currently when a user tries to connect a board via Web Serial, the same dialog box (Select Serial Device - Connect serial device) remains on screen after a user clicks connect and selects their COM port. This is confusing to users because they aren't sure if a connection is in progress or has failed. The user can also keep clicking the same series of prompts with no feedback. I think it would be helpful to let a user know that the editor is trying to connect to their board that they selected or to tell them that a connection failed.

makermelissa, 2024-08-26:
Yep, I'm running into this with the C6. It's not getting the prompt info back from the board. I think this is the first thing to fix.

makermelissa, 2025-02-07:
Rather than having a new dialog appear, I went with the approach of showing it as busy. This should remove the confusion.

PR #292 (2025-02-08): Show busy while user selects port and handle canceling properly
Fixes #240.