circuitpython/web-editor #244: Serial Plotter?

FoamyGuy, 2024-08-29:
Is there interest in having a serial plotter similar to the one inside of Mu?

makermelissa, 2024-08-29:
Possibly. I believe I did one in another project (https://github.com/adafruit/Adafruit_WebSerial_Plotter) that could possibly be ported here.

FoamyGuy, 2024-08-29:
Nice, thank you for the link, I hadn't seen that before. 

I started tinkering with this idea a bit this morning, I've gotten  as far as making a spot in the layout for it, and a button that reveals/hides it but no actual plotting functionality yet. I'm planning initially on putting it in the lighter grey box within the serial-page portion of the layout, but am open to feedback or other suggestions if you have different ideas about how or where to integrate it into the UI.

[image]

I've used Apex Charts (https://apexcharts.com/) for a few things recently and was thinking of using it for this plotter. But I have also used Chart.js in the past which it looks like the older serial plotter you worked on uses. I may opt for Chart.js instead and see how much of that older version I can re-use. 

I'm assuming that the XT and XY plot types are like "value over time" vs. "value over other value". I hadn't considered that option before but it does seem like nice functionality. 

The buffer size is nice as well, I thought about it some but hadn't settled on any solution beyond having the clear button also clear out the chart data in addition to the terminal output.

makermelissa, 2025-06-23:
Fixed by #250 and #314.

PR #250 (2024-09-11): Serial plotter Implementation
resolves: #244 

Supersedes: #245 which I accidentally merged the wrong branch to and ended up with failing CI + some changes that were unrelated to the plotter.

Screenshot: 
[image]

Automatic parsing of printed lines support the following formats:

- single value i.e. `42.92`
- comma seperated values i.e. `42.92, 21.29, -18.13`
- Values in a list i.e. `[42.92, 21.29, -18.13]`
- Values in a tuple i.e. `(42.92, 21.29, -18.13)`

I have tested only USB workflow and testing was performed on a feather esp32-s3 TFT with an I2C accelerometer attached.
closed 2025-06-23 by makermelissa