circuitpython/web-editor #266: Support for custom hostname in settings.toml

jdimpson, 2024-11-15:
Would like a way to do the equivalent of setting `wifi.radio.hostname="newhostname"` prior to calling `wifi.radio.connect()`, but via `settings.toml` so that the CPy dhcp client requests "newhostname" along with requesting an IP address. This way it would take effect when using web workflow.

I tried assigning to CIRCUITPY_WEB_INSTANCE_NAME but that didn't seem to affect what hostname the dhcp client requests. Presumably something like CIRCUITPY_WIFI_HOSTNAME would be more appropriate.

dhalbert, 2024-11-15:
Could you open this in https://github.com/adafruit/circuitpython instead? It's a core CircuitPython issue, not the web editor. Thanks.

makermelissa, 2025-02-06:
Opened in the appropriate repo as https://github.com/adafruit/circuitpython/issues/10033, so closing this one.

closed 2025-02-06 by makermelissa