circuitpython/web-editor #289: serial status line missing on beta

FoamyGuy, 2025-01-28:
### CircuitPython Version

```python
Adafruit CircuitPython 9.2.3 on 2025-01-17; Adafruit Sparkle Motion with ESP32
```

### Workflow(s)

USB

### Browser(s)

```python
Google Chrome 132.0.6834.110
```

### Console Log

```python

```

### Steps

connect board with USB workflow

open serial pane

### Description

Live version:
[image]

Beta version:
[image]

### Additional information

The status information does blink visible momentarily I think. Probably something about the plotter button has broken it.

FoamyGuy, 2025-01-29:
Maybe Wifi Off vs. On plays into this somehow. Currently I am seeing the status info on the beta page. My wifi is currently off. 

I'm not positive what state it was in when the issue was created and the screenshot showing it missing was taken, but it's possible that it was on for that screenshot.

makermelissa, 2025-02-06:
I just tried beta and it shows for me. I suspect the issue was actually the new plotter button made the contents of the serial pane a bit wider, which makes the status disappear due to not having enough room.

makermelissa, 2025-02-11:
Closing as this is the intended behavior.

closed 2025-02-11 by makermelissa