circuitpython/web-editor #290: USB REPL File Mode - Download Files Not Working

FoamyGuy, 2025-01-29:
### CircuitPython Version

```python
Adafruit CircuitPython 9.2.3 on 2025-01-17; Adafruit Sparkle Motion with ESP32
```

### Workflow(s)

USB

### Browser(s)

```python
Chrome Version 132.0.6834.110 (Official Build) (64-bit)
```

### Console Log

```python
index.js:20 Uncaught (in promise) TypeError: Failed to execute 'createObjectURL' on 'URL': Overload resolution failed.
    at Function.l [as saveAs] (index.js:20:20083)
    at Pd._download (index.js:31:101955)
    at async Pd._handleDownloadButton (index.js:31:101145)
```

### Steps

1. connect to board with USB workflow
2. press open button
3. select a single file
4. press download button

### Description

Download does not complete and error is printed in console. 

### Additional information

If you select multiple files it does not have the same error in the console but the resulting downloaded zip contains 0 byte files.

makermelissa, 2025-01-31:
Since you are using the ESP32, which goes through the console, this may be related to #289.

makermelissa, 2025-02-11:
Ok, now that that's done, I tried it again on beta. It seems to be failing with reading the file. After some investigating, I notice (even testing locally) that it works fine with my local copy of circuitpython-repl-js, but not when using the published one despite it being up to date.

makermelissa, 2025-02-11:
Blowing away my `node_modules` folder and reinstalling seems to have fixed it.

makermelissa, 2025-06-23:
Fixed by #291 and #314.

PR #291 (2025-02-11): Fix file download issues
Fixes #290. This also updates the version of circuitpython-repl-js with some fixes in it that were added in https://github.com/adafruit/circuitpython-repl-js/pull/15.
closed 2025-06-23 by makermelissa