circuitpython/web-editor #312: Code Editor has trouble handling byte strings.

phesterman, 2025-05-12:
### CircuitPython Version

```python
Adafruit CircuitPython 9.2.7 on 2025-04-01; 01Space 0.42 OLED ESP32C3 with ESP32-C3FH4
```

### Workflow(s)

USB

### Browser(s)

```python
Google Chrome Version 136.0.7103.93 (Official Build) (arm64) on MacOS Sequoia 15.4.1
```

### Console Log

```python

```

### Steps

The code editor seems to garble byte strings, so that something like this (that works in Thonny) generates an error:

peer = espnow.Peer(mac=b'\x18\x37\x2F\x52\x7C\x7C') [this is a made-up MAC address]

Save + Run produces "ValueError: buffer length must be 6" on that line.  After closing and reopening the file, I discovered that the code editor has changed the line to:

peer = espnow.Peer(mac=b'00ù3È') [this is the actual line from the code editor]

makermelissa, 2025-05-13:
Thanks for submitting this issue. It appears to be converting the character bytes into their constituent characters rather than leaving them in their native state. It could simply be an issue with properly handling backslashes when saving.

makermelissa, 2025-06-23:
I was una ble to replicate the issue. However, I have #314 open to merge the beta code into main and that included a release to fix backslashes in F-strings.

makermelissa, 2025-06-23:
Closing for now. Feel free to re-open if you experience this again.

closed 2025-06-23 by makermelissa