circuitpython/web-editor #313: Bluetooth File Transfer not working on ESP32-S3 (YD-ESP32-S3-N16R8)

maxoum-shi7su, 2025-05-22:
### CircuitPython Version

```python
CircuitPython versions tested:

8.2.7 (fr)

9.2.7

10.0.0-alpha.6
```

### Workflow(s)

BLE

### Browser(s)

```python
Google Chrome 136.0.7103.114
On macbook pro m2, linux and windows
```

### Console Log

```python
no error
```

### Steps

1. Flash the board with one of the following CircuitPython versions:
   - `adafruit-circuitpython-yd_esp32_s3_n16r8-fr-8.2.7.bin`
   - or `9.2.7`, or `10.0.0-alpha.6`
2. After reboot, wait for the safe mode LED pattern, then **press the BOOT button** to enable the BLE workflow.
3. Go to [https://code.circuitpython.org/](https://code.circuitpython.org/)
4. Click on **Bluetooth** to connect via Web Bluetooth.
5. Select the ESP32-S3 board from the list of available devices.
6. Observe the following:
   - The connection seems to succeed (no error message),
   - No files are listed on the board,
   - Attempting to upload a file shows the loading animation (the snake) indefinitely, but the upload never completes.

### Additional information

I also tested [File Glider](https://learn.adafruit.com/file-glider): the board appears, but selecting it throws an error.

Web Serial works correctly with the board.

Unfortunately, I don’t have an nRF52840 board to compare against.

makermelissa, 2025-06-23:
I had made some improvements in Beta. I just merged all the beta code into main, so please give it another try.

MaximeMauduit, 2025-06-25:
I still have the same problem, tried to dig into the code but everything seems fine no error anywhere.

makermelissa, 2025-06-25:
Ok, thanks for checking. I'll keep this open.